CommonWealthRobotics/BowlerStudio #424: Fails in CI to import via ScriptingEngine

JansenSmith, 2024-09-09:
[Failed CI build](https://github.com/JansenSmith/SecretMessageCoin/actions/runs/10778396009/job/29890213226)
```
File   SecretMessageCoin.groovy
Current branch is 6f2a5ba65e4aeae4653263a576233fa19007a07a need main
Getting branches https://github.com/JansenSmith/SecretMessageCoin  
Found upstream refs/heads/main
java.lang.RuntimeException: java.lang.RuntimeException: org.eclipse.jgit.api.errors.RefNotFoundException: Ref origin/main cannot be resolved
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.cloneRepo(ScriptingEngine.java:1752)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.fileFromGit(ScriptingEngine.java:1095)
	at org.codehaus.groovy.vmplugin.v8.IndyInterface.fromCache(IndyInterface.java:321)
	at SecretMessageCoin.run(SecretMessageCoin.groovy:30)
	at com.neuronrobotics.bowlerstudio.scripting.GroovyHelper.inline(GroovyHelper.java:58)
	at com.neuronrobotics.bowlerstudio.scripting.GroovyHelper.inlineScriptRun(GroovyHelper.java:78)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.inlineScriptRun(ScriptingEngine.java:511)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.inlineFileScriptRun(ScriptingEngine.java:1068)
	at com.neuronrobotics.bowlerstudio.BowlerKernel.runArgumentsAfterStartup(BowlerKernel.java:214)
	at com.neuronrobotics.bowlerstudio.BowlerKernel.main(BowlerKernel.java:123)
Caused by: java.lang.RuntimeException: org.eclipse.jgit.api.errors.RefNotFoundException: Ref origin/main cannot be resolved
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.checkout(ScriptingEngine.java:1623)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.cloneRepo(ScriptingEngine.java:1750)
	... 9 more
Caused by: org.eclipse.jgit.api.errors.RefNotFoundException: Ref origin/main cannot be resolved
	at org.eclipse.jgit.api.CreateBranchCommand.getStartPointObjectId(CreateBranchCommand.java:246)
	at org.eclipse.jgit.api.CreateBranchCommand.call(CreateBranchCommand.java:99)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.shallowCheckout(ScriptingEngine.java:1640)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.checkout(ScriptingEngine.java:1583)
	... 10 more
Usage: 
java -jar BowlerScriptKernel.jar -s <file 1> .. <file n> # This will load one script after the next 
java -jar BowlerScriptKernel.jar -p <file 1> .. <file n> # This will load one script then take the list of objects returned and pss them to the next script as its 'args' variable 
java -jar BowlerScriptKernel.jar -r <Groovy Jython or Clojure> (Optional)(-s or -p)<file 1> .. <file n> # This will start a shell in the requested langauge and run the files provided. 
java -jar BowlerScriptKernel.jar -g <Git repo> <Git file> # this will run a file from git
QuantumRenderer: shutdown
Error: Process completed with exit code 1.
```

madhephaestus, 2024-09-10:
change this https://github.com/JansenSmith/SecretMessageCoin/blob/4844ac9721cf50e622716c3773e661b979175b73/SecretMessageCoin.groovy#L33C1-L33C18 to null

CI checkouts of the code are incomplete, just containing the commit being worked on, so there are no branched in the CI checkout. This is an inconsistency in how BowlerStudio clones a repo and how CI "clones" a repo.

madhephaestus, 2024-09-10:
I pushed a change to the latest kernel, try re-running the existing CI to see if it completes as expected now?

closed 2024-09-10 by madhephaestus

madhephaestus, 2024-09-12:
did this work?