CommonWealthRobotics/BowlerStudio #426: Mac Issue

KacperLa, 2024-10-18:
Per your request :D

java.lang.reflect.InvocationTargetException
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at com.sun.javafx.application.LauncherImpl.launchApplicationWithArgs(LauncherImpl.java:389)
	at com.sun.javafx.application.LauncherImpl.launchApplication(LauncherImpl.java:328)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at sun.launcher.LauncherHelper$FXHelper.main(LauncherHelper.java:893)
Caused by: java.lang.RuntimeException: java.io.IOException: Need authenticator
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.setWorkspace(ScriptingEngine.java:214)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.getWorkspace(ScriptingEngine.java:569)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.getRepositoryCloneDirectory(ScriptingEngine.java:2259)
	at com.neuronrobotics.bowlerstudio.scripting.ScriptingEngine.cloneRepo(ScriptingEngine.java:1711)
	at com.neuronrobotics.bowlerstudio.BowlerStudio.main(BowlerStudio.java:279)
	... 11 more
Caused by: java.io.IOException: Need authenticator
	at sun.nio.ch.FileDispatcherImpl.read0(Native Method)
	at sun.nio.ch.FileDispatcherImpl.read(FileDispatcherImpl.java:46)
	at sun.nio.ch.IOUtil.readIntoNativeBuffer(IOUtil.java:223)
	at sun.nio.ch.IOUtil.read(IOUtil.java:197)
	at sun.nio.ch.FileChannelImpl.read(FileChannelImpl.java:159)
	at sun.nio.ch.ChannelInputStream.read(ChannelInputStream.java:65)
	at sun.nio.ch.ChannelInputStream.read(ChannelInputStream.java:109)
	at sun.nio.ch.ChannelInputStream.read(ChannelInputStream.java:103)
	at sun.nio.cs.StreamDecoder.readBytes(StreamDecoder.java:284)
	at sun.nio.cs.StreamDecoder.implRead(StreamDecoder.java:326)
	at sun.nio.cs.StreamDecoder.read(StreamDecoder.java:178)
	at java.io.InputStreamReader.read(InputStreamReader.java:184)
	at java.io.BufferedReader.fill(BufferedReader.java:161)
LatestFromGithubLaunch clean exit

madhephaestus, 2024-10-20:
This issue was resolved by moving the working directory from ~/Documents/bowler-workspace to ~/Library/Application Support//BowlerLauncher/ on MacOS

closed 2024-10-20 by madhephaestus

madhephaestus, 2024-10-22:
BowlerLauncher should now work from the auto-updater again