CommonWealthRobotics/BowlerStudio #431: integrate PyMeshLab for meshing_repair_non_manifold_vertices

JansenSmith, 2024-11-08:
ms.meshing_repair_non_manifold_vertices(vertdispratio = 0.010000)
running this twice fixes most STLs with duplicated vertices 

Here is a clean pr adding SceneBuilder as an example of all the bits that need to be touched:
https://github.com/CommonWealthRobotics/BowlerStudio/issues/409

Process is documented here: https://github.com/CommonWealthRobotics/BowlerStudio/pull/412

MeshLab is a utility, so it would just require an additional json in ExternalEditorsBowlerStudio and a static class call in:
https://github.com/CommonWealthRobotics/bowler-script-kernel/blob/kh/add-cadoodle/src/main/java/com/neuronrobotics/bowlerstudio/util/GeometrySimplification.java

JansenSmith, 2024-11-08:
actually..... i've found some that require multiple runs of meshing_repair_non_manifold_vertices....
instead, use PyMeshLab to find non-manifold vertices and iterate until number is zero.

JansenSmith, 2024-11-09:
actually..... i've found some that require multiple runs of meshing_repair_non_manifold_vertices....
still researching

madhephaestus, 2024-11-10:
You may want to look at the source of the exception you are trying to correct and see if maybe you can avoid the problem upstream of the repair: https://github.com/NeuronRobotics/JCSG/blob/development/src/main/java/eu/mihosoft/vrl/v3d/Plane.java#L106

madhephaestus, 2024-11-10:
I need a small file <1mb that produces the problem for unit testing

madhephaestus, 2024-11-10:
the root cause may have been fixed in this commit: https://github.com/NeuronRobotics/JCSG/commit/5974f4a3f041e91cd2872f5474e0fdba1337a0f3